jnilo1/rtl8196e-gateway #11: [BUG] No update + lost root password

jeanbruder, 2025-04-07:
Dear J Nilo,

Here's what happened to me.

Step 1 : Execute the script.

./flash_ezsp13.sh zbgw02 ncp-uart-hw-7.4.5.gbl 
root@zbgw02's password: 
........Starting firmware transfer
Sending /tmp/firmware.gbl, 1608 blocks: Give your local XMODEM receive command now.

... Nothing else after a time ... Keyboard ENTER

Xmodem sectors/kbytes sent:   0/ 0k

... Nothing else after a time ... Keyboard break (CTRL+C)

Step 2 : SSH to the remote box.

ssh root@zbgw02
root@zbgw02's password: 
Permission denied, please try again.
root@zbgw02's password: 

Z2M can connect to the box :

[image]

Can you please help in :
- updating the firmware,
- recovering/chaging the root password.

Best regards,
Jean

jnilo1, 2025-04-08:
@jeanbruder 
1/ What is the current version of the firmware installed in your gateway ? where did you get it from ?
2/ ssh apparently worked ok in the failed attempt to update the firmware. So you entered the good password. The flash script does not modilfy the password s! Reboot your gateway and see I you can ssh again
Jacques

jnilo1, 2025-04-08:
Also please use the discussions tab for help support so anyone can jump in :-)

jnilo1, 2025-04-09:
@jeanbruder Please check https://github.com/jnilo1/hacking-lidl-silvercrest-gateway/discussions/15

jnilo1, 2025-04-09:
Issue closed. Probably not a bug.

closed 2025-04-09 by jnilo1