jnilo1/rtl8196e-gateway #25: [BUG] firmware update script does not work in new root filesystem

hlyi, 2025-06-29:
## Describe the bug
After flash new root filesystem, the firmware update no longer work.

I think there are two issues:
1) tar in new root filesystem doesn't support compression. So when untaring in new root filesystem, the following error occurs:
`tar: invalid tar magic`
2) `sx` depends on `/lib/ld-uClibc.so.0`, but the `ld-uClibc.so.0` doesn't exist in new root filesystem, so when the following error occurs:
`-sh: /tmp/sx: not found`

Could you please provide a statically linked 'sx`?

jnilo1, 2025-06-30:
Check here:
https://github.com/jnilo1/hacking-lidl-silvercrest-gateway/tree/main/1-Firmwares/14-NCP-UART-HW/firmware

zelmis, 2025-06-30:
$ ./flash_ezsp8.sh 192.168.100.147 ncp-uart-hw-7.5.0-btlhw.gbl
root@192.168.100.147's password:
tar: invalid tar magic
chmod: sx: No such file or directory