jnilo1/rtl8196e-gateway #28: [BUG] Device reguarly timeouts with respect to Zigbee2MQTT

skinkie, 2025-07-25:
## Describe the bug

The device reguarly timeouts with respect to Zigbee2MQTT. Herstartin Zigbee2MQTT does not help, the device itself needs to be rebooted.

I will check if I can see anything over SSH, but last time I checked it, also the network was not available anymore. One green light is lit, switch suggests network link with bridge.

## To Reproduce

Not consistently.

## Expected behavior

The devices should stay available.

## Environment

- Gateway Model: have to check
- Firmware version: latest from the repo
- Zigbee stack (EZSP version, RCP, etc.): 13
- Host software (Z2M, ZHA, ...): Zigbee2MQTT (lately running latest dev)

jnilo1, 2025-07-26:
I have been using the firmware 7.5.0 from the repo for nearly 60 days without any problem:
```
Version Zigbee2MQTT                                         2.2.1 commit: a14feef8c13f5ce49b0ac358d98d3234d21d7ff4
Type de coordinateur                                        EmberZNet
Version du coordinateur                                     7.5.0 [GA]
```
So I don’t believe this is a firmware bug — I imagine many users would have reported problems by now if it were.

From my experience, Zigbee2MQTT timeouts are often related to poor Ethernet connection quality and/or unstable switches.

As a diagnostic step, try connecting your Lidl gateway directly to your Zigbee2MQTT host using a short Ethernet cable (without any router or switch in between). Assign static IPs manually.

To maintain Internet access during the test, you can use Wi-Fi or plug in a USB-to-Ethernet adapter on the host  to get a second network interface

This setup isolates the Zigbee gateway from potential issues in your LAN infrastructure. If the timeouts disappear in this direct connection mode, your problem is likely due to poor switch or cable quality.

skinkie, 2025-07-26:
I'll follow up when I have more details, sinds a few weeks it happens every 2-3 days, without any change in infrastructure.

skinkie, 2025-07-29:
Today: timeout Zigbee2MQTT, ssh still reachable, restarting Zigbee2MQTT works

skinkie, 2025-08-05:
Ok so today, another event. I can sign in to ssh. But Zigbee2MQTT does not want to reconnect.

```
Starting Zigbee2MQTT without watchdog.
[2025-08-05 11:29:17] info:     z2m: Logging to console, file (filename: log.log)
[2025-08-05 11:29:17] info:     z2m: Starting Zigbee2MQTT version 2.5.1-dev (commit #713551b5)
[2025-08-05 11:29:17] info:     z2m: Starting zigbee-herdsman (5.0.0)
[2025-08-05 11:29:17] info:     zh:ember: Using default stack config.
[2025-08-05 11:29:17] info:     zh:ember: ======== Ember Adapter Starting ========
[2025-08-05 11:29:17] info:     zh:ember:ezsp: ======== EZSP starting ========
[2025-08-05 11:29:17] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-08-05 11:29:17] info:     zh:ember:uart:ash: Socket ready
[2025-08-05 11:29:17] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-08-05 11:29:22] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-08-05 11:29:22] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-08-05 11:29:27] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-08-05 11:29:27] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-08-05 11:29:32] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-08-05 11:29:32] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-08-05 11:29:37] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-08-05 11:29:37] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-08-05 11:29:42] error:    z2m: Error while starting zigbee-herdsman
[2025-08-05 11:29:42] error:    z2m: Failed to start zigbee-herdsman
[2025-08-05 11:29:42] error:    z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-08-05 11:29:42] error:    z2m: Exiting...
[2025-08-05 11:29:42] error:    z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@5.0.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:665:19)
    at EmberAdapter.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@5.0.0/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1547:24)
    at Controller.start (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman@5.0.0/node_modules/zigbee-herdsman/src/controller/controller.ts:134:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:67:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:101:13)
    at start (/opt/zigbee2mqtt/index.js:149:5)
 ELIFECYCLE  Command failed with exit code 1.
```

De last part of dmesg is:
```
Freeing unused kernel memory: 192K (802f0000 - 80320000)
cmd read

b8000038: 2794A104  0000000F    00000042  00000018    '��        B    
cmd write
Write memory 0xb8000038 dat 0x1794a104: 0x1794a104
```

Restarting tuya_gateway does not help. After the connection I get: 
```
Connect from host 192.168.5.254 fd=5
Closing existing connection
```

When I reboot the box, zigbee2mqtt works again.

skinkie, 2025-08-07:
I have only flashed the zigbee part, not the rootfs. My power is provided by a very short USB-cable. I can connect something like a USB power meter. But since the rest of the gateway works, I am puzzled. Maybe I should implement a reboot command when zigebee2mqtt quits.

skinkie, 2025-08-16:
@jnilo1 I have  channel set to 14 in the configuration. WiFi is 5-13 (hence Zigbee 15 and higher).

closed 2025-12-19 by jnilo1